bjoernkarmann/project_alias #9: Offer.

Match-it, 2019-01-15:
Apologies if this isn't allowed. I'm new. I wanted to declare that I would pay someone to put this together for me. I don't have the technical know-how to do it, or the resources to try.

sempei13, 2019-01-19:
I’d be willing to do it if there are no objections.

bjoernkarmann, 2019-01-21:
No objections here. If anyone needs help make an Alias they can, of course, get someone to create it for them. Just be aware that Alias in the current state is not to compare with commercial out-of-the-box experience. Even when built, it may take some tweaks to get it working properly. And you will have to update the software, once we update the source code.

closed 2019-01-21 by bjoernkarmann

retorquere, 2020-02-15:
@sempei13 I'm interested. I don't mind software updates later, I'm a software engineer, but I don't have access to a 3D printer and I am no good with a soldering iron.