bjoernkarmann/project_alias #13: Speaker Wiring to JST2.0 Speaker Out on ReSpeaker?

pierow2k, 2019-01-19:
There are a single pair of red and black wires connected to the `JST2.0 Speaker Out` on the ReSpeaker board and two pairs of red and blue wires connected to the speakers.  

Are the red & black wires broken out to a parallel connection to the speakers under the board?  _(I am assuming that since these are 50 ohm speakers they are wired in parallel and not in series)_

[image]

esmsnt, 2019-01-19:
I put one of these together last night, I had difficulty finding 50ohm speakers in the form factor needed, but there are a few 8ohm options roughly 16mm in size on digi-key and others.  I have a pair of SP-1605s in series right now working fine at the default 18% volume output from the ReSpeaker hat.  The ReSpeaker hat JST is also not labeled for polarity, just choose one, should be fine.

Word of advice, the wires on the SP-1605 are tiny, I would recommend actually using something like AS01808MR-LW152-R, have two of these sitting here too, the speakers are a little larger but the leads are much easier to work with.

As far as the wiring goes, as long as you keep the speakers in the same phase it should be fine.  Impedance match would be nice but the power levels are low and you don't care about maximum power transfer so I don't think it matters.

```
Serial Wiring:
JST(+) ------ (+) Speaker A (-) ------ (+) Speaker B (-) ------ (-)JST

Parallel Wiring:
JST(+) ------ < ------ (+) Speaker A (-) ------ > ------ (-)JST
              | ------ (+) Speaker B (-) ------ |  
```

pierow2k, 2019-01-20:
Thanks for the reply.  I had some inexpensive 26mm 8 ohm speakers on-hand and ran them in parallel.  I couldn't find anything specific in the ReSpeaker docs regarding impedance but I assume that a 4 ohm load won't present a problem.  

[image]

closed 2019-01-20 by pierow2k