bjoernkarmann/project_alias #14: Memory Error when trying to install tensorflow

LazyG, 2019-01-20:
For reference in case others experience the same issue I did when trying to install on a 2018-11-13-raspbian-stretch-full.img version of Raspian.

When trying to install tensorflow via command: sudo pip3 install tensorflow keras, I received the following memory error.

[image]

However, when I changed the command to the following it resolved the issue.
pip3 install --no-cache-dir tensorflow

bjoernkarmann, 2019-01-21:
`pip3 install --no-cache-dir tensorflow` fixes this issue. To find more on related issues please refer to https://github.com/tensorflow/tensorflow/issues.

closed 2019-01-21 by bjoernkarmann

rahulshyokand, 2020-08-11:
> `pip3 install --no-cache-dir tensorflow` fixes this issue. To find more on related issues please refer to https://github.com/tensorflow/tensorflow/issues.

Worked!

shinkai-technology, 2021-06-22:
> `pip3 install --no-cache-dir tensorflow` fixes this issue. To find more on related issues please refer to https://github.com/tensorflow/tensorflow/issues.
Didn't work for me, but thanks

tsei902, 2022-05-04:
worked as well for me!