bjoernkarmann/project_alias #15: Alis Triggers for any noise

helixios, 2019-01-20:
I've completed the build, but at the moment, once trained, the 'Alexa' command is triggered for any noise at all, even if I knock on the table next to it. I'll try a fresh install and post an update, but wondered whether this issue had been experienced before?

esmsnt, 2019-01-20:
I just completed this recently too, having a similar issue, it triggers for almost any speech or conversation.  Additionally, Alexa's response sometimes also triggers the alias wake word again and then interrupts Alexa's ongoing response.

I am guessing more training is needed, I am going to start digging into it this week.

torekndsn, 2019-01-21:
We are still trying to find the best way to create a model that can work “out of the box”. So far our proof of concept has been created around small models trained on the spot for both the wake-word class and the neutral class. While it has been possible for us to create a ok reliable system in the setting it has been trained in (for both wake-word and background sounds), it often fails when brought into a new setting, due to the changes in acoustic, voice, etc. 

The first thing you can try is to record more examples of background sounds that it should NOT react on. Different voices and ambient noise can have an effect on the accuracy of the model. In the interface of the app, you can toggle the background training to ON and add some ambient sounds and words you found triggered it to the pool of examples. After that press train and it will load the new background samples through the NN. We have also experienced that it’s possible to overtrained the wake-word. We have found that just a few examples are enough (around 5-8). 

To make it a bit easier you can record the background examples as wav files with this tool: https://github.com/torekndsn/Dataset_generator-ProjectAlias

Included is a tool to convert the wav files into datasets for the background sounds and save them as numpy arrays. These arrays can be uploaded into the data folder of in Project Alias, and load them inside ai.py line 37 and 38.

helixios, 2019-01-21:
I’ll try on the weekend and let you know how I get on

ChrisVeigl, 2019-01-27:
same problem here. triggers for any kind of noise - e.g. tipping / scratching with fingers next to the mic. 
environmental condition / background noise same like in the recording of the hotword samples. 

can the dataset generator also be used for the hotword samples (it might be helpful to use the same mic for getting the test data) ?

BTW: during training, a lot of "out of memory" messages are thrown - could that be related to the issue ?

torekndsn, 2019-01-28:
> same problem here. triggers for any kind of noise - e.g. tipping / scratching with fingers next to the mic.
> environmental condition / background noise same like in the recording of the hotword samples.
> 

Did you try to create your own dataset for the background noises? And how many examples did it contain?

> can the dataset generator also be used for the hotword samples (it might be helpful to use the same mic for getting the test data) ?
> 

 The dataset generator can also be used for the hotword, by changing the label to `([1.,1.])` instead of`([1.,0.])` in line 83 in datacreater.py
`labels = np.repeat(np.array([1., 0.])[None, :], num_of_spec, axis=0)` You might need to make a numpy file for both the wake-word examples, and the background/noise examples and merge them into on array before training.

> BTW: during training, a lot of "out of memory" messages are thrown - could that be related to the issue ?
>
I dont recall experiencing this.. I'll give it a look in one of the following days.

ChrisVeigl, 2019-01-28:
thanks for the hint!

i trained 5 examples for background noise - but not the tipping / scratching. however it is difficult to anticipate background sounds that could happen when using the device in a "real-life" situation ...

rttgnck, 2019-01-30:
I haven't done any training myself. If I did it would be my first venture. But as I sit here setting a unit it up while watching a movie I wondered if it would be worth gathering some data with one running in front of a loop TV with a loop of movies/tv shows? Would that be a good kind of background noise to filter out false triggers?

bartbergman, 2019-02-27:
Same issue here.
It seems to be triggered by anything. even complete silence triggers... basically it is saying "Ok Google" on and on...
Already did a "reset Alias" a few times, recorded 5 times the keyword, 10 times as well. Also added background noises.
All without success.

lord-x-type, 2019-03-02:
Same here... 
i tried to make different background noise samples and tried different alias names... all without luck.
I am playing around with this issues for 4 days now. I'm using an other sound-card than the suggested respeaker 2 mic. Could this be a problem? i can record and play sound with this card.. no problem.
I'm using the mic card form raspiaudio. My new echo-dot 3 is originally sealed beside me since 3 days now. I only want to use it, if the project_alias is working for me :) sorry for my english.. i'm no native speaker.

lord-x-type, 2019-03-08:
Hi everyone... today my respeaker 2 mic pi hat arrived.
First try: nearly everything is working.. but that is an other issue.

warriar, 2019-06-22:
For me it was the same however it hink this is because it learns the noise!

Try turning down the speakers before learning the trigger word with alsamixer!

philg1234, 2019-09-25:
Hi everyone, 
Is there any news on this topic and the frequent triggering? 

Have you found a way to learn the best? A question too to the background noise. 

How exactly should these be best absorbed?

Greetings phil

philg1234, 2019-09-28:
Today I have tried again a lot to get the project alias a new name.

However, it does not work.

I reset everything via "Reset Alias" and initially learned the alias without any background sound.

As soon as I have learned this, alias will start every 2 seconds without anything being said.

When I learn the alias with the background sound ON, the alias does not start, no matter what is said.

Does this project work for anyone?

Does anyone know about the Verison 2.0?

philg1234, 2019-09-29:
I have spent another hour now with tests, which are the best settings in Alsamixer.

If the default settings of the MICs are used, Alias ​​will start with me every second.

I have a micro completely turned off and the second on the level 32 (lowest) stand.

Now I have fewer problems with the jump of the Alexa.

I learned the word about 20 times and I have to say that it works reasonably well.

I just do not understand correctly how I have to learn the background.

If background mode is ON, do I only need to record backgrounds, or do I have to teach the keyword as well?

What is also unclear to me, whether one function overwrites the other.

Does that mean that if I do train alias without a background, is it gone and vice versa?

Or is that complementary?

It would be cool if someone can explain their attitudes if it works on someone.

Greetings Phil

torekndsn, 2019-09-29:
Hej philg1234, 

Thanks for experimenting and pointing out these questions!  

When background mode is turned ON, it means that all examples recorded should be what Alias should NOT care about. So if you find Alias to be triggered by mistake or by silent, then it should be to record more examples of silent or whatever sound that triggers Alias by mistake. Personly I found it to work best with a 1:5 relation between wake word examples and background noise examples. 

Unfortunately, it takes a lot of experimentation to find a good balance and working setting, which is why we started working on Version 2 to make all this much easier. Several things have changed, and especially the training is much easier now, and no background training is required. We hope to share this within the next week!

philg1234, 2019-09-29:
Hi torekndsn,

Thanks for your quick reply.

All right, then I'll experiment a little bit again.

I'm looking forward to the version 2.0, because I like this project, I think it's pretty cool.

strick213, 2019-12-28:
I finally got it working but, like others, it constantly says "Alexa". Any news on when 2.0 will be coming out? I really appreciate your hard work and ingenuity. This is such a great idea!

torekndsn, 2020-01-05:
Hi, we are glad to announce Alias 2.0 is now available!! We are well aware that this comes very late and has been requested by many who struggled to install or even calibrate Alias. We are sorry about that! 

Nevertheless, we are proud of Release 2.0 and have already had a lot of fun with it! We have introduced a bunch of new features that give even more freedom, and the installation process is now made into a single image file. 

Best, Tore

closed 2020-01-05 by torekndsn