bjoernkarmann/project_alias #16: Any modifications required for the Google Home Mini?

donn-smith, 2019-01-24:
I and several coworkers are interested in building these for our personal use. Has anyone tried it with a Google Home Mini as that seems to be the smart speaker of choice in our group? Are the microphones of the mini spaced similarly to the Google Home?

dragon788, 2019-01-26:
With the dome shape on the top of the Google Home Mini the 3D printed part will likely not properly interfere with the microphones to prevent it from hearing things.

dragon788, 2019-01-26:
See https://www.ifixit.com/Teardown/Google+Home+Mini+Teardown/102264 for actual locations.

donn-smith, 2019-01-27:
Then we'll have to develop our own holder. Thanks for the pointers to the teardown. It did confirm where I thought the microphones are.

espoelstra, 2019-01-29:
See also this comparison shot of the mic board of the mini on top of the regular Google Home.

https://drive.google.com/file/d/1y5IdLOxlhdWW_A8DrJrClvi78OX2GwJ6Fw/view?usp=sharing

filirnd, 2019-04-15:
Hi, I'm interessed for the google mini version of Alias. Have you make an opensource 3d model for this ?

donn-smith, 2019-04-15:
I have not and no one has contacted me about doing so. I do not have access to a 3D printer at this time, so I am not pursuing this design right now.

f3rm4rf3r, 2020-11-04:
Any update on this? I am looking for a google home compatible hat. Thanks!

dragon788, 2020-11-04:
The new Nest Mini has a slightly different shape and may have a different mic layout.

Steepy12, 2021-03-06:
The nest mini have 3 mics but alias use only 2 speaker, i got a nest mini i can test to make the 3D file but i need to know if i need to cover the 3 mics with speakers or if 2 speaker will do the job ?

OFFCANNY, 2021-03-12:
> The nest mini have 3 mics but alias use only 2 speaker, i got a nest mini i can test to make the 3D file but i need to know if i need to cover the 3 mics with speakers or if 2 speaker will do the job ?

Hi, I'm also trying to make a tweaked version for Nest Mini. From my research, the Nest Mini also only has 2 speakers. ( https://youtu.be/UdFTvebhk_0?t=178 )

From the info above from other users, I overlayed the Nest Mini on top of the Home, and you can see that the Pi's speakers need to be positioned more inwards. https://postimg.cc/gallery/Z3bpwWY 
[image]
[image]

OFFCANNY, 2021-03-12:
> Hi, I'm also trying to make a tweaked version for Nest Mini. From my research, the Nest Mini also only has 2 speakers. ( https://youtu.be/UdFTvebhk_0?t=178 )
>
> From the info above from other users, I [...]

Correction, the Home Mini has 2 Mics, the NEST MINI does indeed have 3!

OFFCANNY, 2021-03-12:
> The nest mini have 3 mics but alias use only 2 speaker, i got a nest mini i can test to make the 3D file but i need to know if i need to cover the 3 mics with speakers or if 2 speaker will do the job ?
Hey guys, Im very interested in designing a version that will work on the Nest Mini. Lets do this. 
@donn-smith Did you create one?
@Steepy12 Yes it does have 3 mics, we will need to add another speaker output. Also it doesn't
[image]
[image]

 look like it will be that different, the speakers only need to be a few CM inwards. 
[image]

Steepy12, 2021-03-12:
Hum i see,

OFFCANNY, 2021-03-12:
I don't think we need to measure exactly as long as the speaker is somewhat above the mic there should be enough white noise to make it work

OFFCANNY, 2021-03-12:
> Hmm I see, adding the 3rd speaker wouldn't be complicated because the 3d model would be difficult. Especially if there are a few internal cm to know, it would be necessary to be able to dismantle it to measure the distance, or to see to completely redo a new box with the mini nest which would simplify the task of creating the module and of everything integrated in a single box, I will look to make a new box already tested if Alias ​​works and especially to create the 3D model

I don't think we need to measure exactly as long as the speaker is somewhat above the mic there should be enough white noise to make it work

Steepy12, 2021-03-16:
yes it could work, but second problem, how do we so that the sound of the speaker is not blocked by the "shell" which will hold the mini speaker

Steepy12, 2021-03-16:
if @donn-smith made is own file it could works, the nest and the home mini have the same emplacement for the mic on the left and right.

donn-smith, 2021-03-16:
Sorry, I did not. I don't own a 3D printer, so I have not attempted this as of yet.