bjoernkarmann/project_alias #19: Raspberrypi.local not loading on phone

sempei13, 2019-02-13:
I seem to have installed everything fine; no errors.  When I went to train Alias using my phone, raspberrypi.local:5050 never loads.  Any ideas?

sempei13, 2019-02-13:
Oh, and I just tried it on my computer, just to see if it would work, and it says "This site can’t be reached raspberrypi.local refused to connect."  I don't know if that helps or not.

sempei13, 2019-02-13:
More info.  I CAN ping it.  I can also ssh into pi@raspberrypi.local.

sempei13, 2019-02-13:
I got another project to use raspberrypi.local on another port, so it seems like the issue is with Alias.  Like it's not serving the webpage through 5050.

helixios, 2019-02-13:
The application may not be running on startup. Try opening a console, navigating to the project folder and the running the application:
‘python3 app.py’
Then try accessing the page from your phone

sempei13, 2019-02-13:
Looks like something else is going on.  This is what I get: 
Traceback (most recent call last):
  File "app.py", line 8, in <module>
    from modules import connection
  File "/home/pi/project_alias/modules/connection.py", line 7, in <module>
    from modules import sound
  File "/home/pi/project_alias/modules/sound.py", line 4, in <module>
    from modules import led
  File "/home/pi/project_alias/modules/led.py", line 25, in <module>
    LED = Pixels()
  File "/home/pi/project_alias/modules/led.py", line 8, in __init__
    self.dev = apa102.APA102(num_led=self.PIXELS_N)
  File "/home/pi/project_alias/modules/apa102.py", line 93, in __init__
    self.spi.open(bus, device)  # Open SPI port 0, slave device (CS) 1
FileNotFoundError: [Errno 2] No such file or directory

sempei13, 2019-02-13:
I should say, too, that I'm using an AIY kit, not a voice hat, if that makes a difference.  The pi recognizes the kit b/c I can use it for other apps.

sempei13, 2019-02-18:
Figured it out.  I'd skipped part of the installation because I didn't think I needed it, but I did.

closed 2019-02-18 by sempei13