bjoernkarmann/project_alias #21: Model not saved or not activated

sempei13, 2019-02-19:
Sorry to be such a pain.

When I exit out of app.py (it's not automatically starting for me, but that's a different issue, one that I think I can figure out), all is well, until I restart app.py and then it acts as though it's never been trained.  I've tried starting over, having erased the training and that doesn't work.

Any ideas?

SouthernFriedScientist, 2019-02-19:
I'm having the exact same issue. rc.local won't run on start-up, model doesn't retain training between reboots.

sempei13, 2019-02-20:
I'm glad I'm not the only one.

markusche, 2019-02-20:
I have the same two issues too!

sempei13, 2019-02-20:
So, I added this to rc.local: echo "running rc.local" > /tmp/rc_test.txt

Then, I rebooted, so rc.local would have some chance of running.  Once it restarted, I looked at /tmp/rc_test.txt and lo and behold it said "running rc.local" as expected (no quotes).

So, that tells me that it IS running, but not getting to the Alias app or not running it or it's not working.  More testing to do.

sempei13, 2019-02-20:
I thought that maybe the problem was that it wasn't executing on boot, so after figuring out that for some reason, it wasn't executing as a part of rc.local, I put the same line "cd project_alias && python3 app.py &" (without the quotes) into /etc/profile and it executes.  Why it works there, and not in /etc/rc.local, even when other scripts execute both before and after, I don't know.  Permissions issue or something, maybe?!

Turns out, if I restart the pi after training it, the model is either not loaded (it says it is) or not saved (which after training, it says it's saved).  Again, permissions?!

sempei13, 2019-02-23:
I just looked at the permissions for the three models in the data folder.  Mine are set to 644.  I'm going to try and run Alias and check the file sizes before I restart it and after I train it to see if they change from their current size.

sempei13, 2019-02-23:
I trained my models and checked the sizes of the files.  No change.  I even changed permissions from 644 to 755 and that didn't fix it.  Anyone know where the models are saved to when it says "Model saved"?  Because I'm not seeing a current model in the data folder at least.

torekndsn, 2019-02-24:
Thanks for the attention on this issue! 
I found the bug that made the program load an empty model instead of the last saved. The master is now updated. 

The broader issue is that we haven't found a way that makes it possible to load an already trained model and continue adding examples once its loaded. Here is the best work around we have so far that should make it possible to train the model for continuous use: 

1. First change line 29 in ai.py to LOAD_MODEL = False. 
2. Then change line 15 in globals.py to UPDATE_BG_DATA = True
This creates new models and empty numpy arrays to store the examples. (This will override existing saved numpy arrays) 

3. Launch the app and record the needed background examples for the null state. around 100-200 examples should be enough while trying to cover different sounds, words, and conversations. Dont add examples for the active state yet.
4. Train the model, as this will save a new neutral model (See line 111 in ai.py) and save the numpy arrays containing the background examples (See line 112-113 in ai.py).
5. Now change the booleans in step 1 and 2 back so that line 29 in ai.py is set to LOAD_MODEL = True and line 15 in globals.py is set back to UPDATE_BG_DATA = False. 
6. Launch the app again and the record background examples from before should be loaded. Now train a new wake-word and train the app. Now it should work! 

It's not the most convenient way, and we should find a better way to get around this! 
Thanks again for finding this issue.

patbidouille, 2019-02-24:
for rc.local, the service does not create under Debian Stretch. But this is possible:
google for a doc
or
https://www.apb-informatique.fr/debian-9-stretch-rc-local-obsolete/
(in French, sorry)

I mounted the project and launched it directly:
cd project_alias && python3 app.py &
I have a lot of noise and the link rasp: 5050 works.
By cons if I say "alias" the noise does not stop.
And I do not understand the trainning function well, it is to learn a background noise? or a noise / activation word?
Last point, if we launch training, it displays training then points, but impossible to stop it.
More details would be "top", thank you

sempei13, 2019-02-25:
So for clarification, once you change those two lines in those two files, train the background sounds, change the two lines back, and then train the wake word, can you then train background sounds again so that it adds to the possible background sounds, or do you need to either start over, or do you need to change the files again?

torekndsn, 2019-02-25:
> So for clarification, once you change those two lines in those two files, train the background sounds, change the two lines back, and then train the wake word, can you then train background sounds [...]

Correct, and yes, you can update the background sounds, by only changing the UPDATE_BG_DATA back to True. Then it will still load the existing examples and add any new recorded background sounds on top of it. When UPDATE_BG_DATA = True, the app goes into a state, that only serves to update the background examples arrays that gets saved locally. Also note that it's not possible to record wake-word examples here. 

When you are done adding new BG examples, set UPDATE_BG_DATA back to False, launch the app and train your wake word.

lord-x-type, 2019-03-10:
Sorry.. I did not get it. Everytime i reboot the device my wakeword has to be trained again. What am i doing wrong? Everything else is working now. :)

sempei13, 2019-03-11:
Did you do what torekndsn said above?

lord-x-type, 2019-03-11:
Hi, sorry for the confusion: my problem was, when i restarted the app and connected with my browser to port :5050 he said examples "0" and not "say your alias"... that mad me wonder an i thought my model with the wakeword was not loaded... but it actually was. problem solved. thanks a lot for this program.

honk0504, 2019-03-30:
> I thought that maybe the problem was that it wasn't executing on boot, so after figuring out that for some reason, it wasn't executing as a part of rc.local, I put the same line "cd project_alias && [...]

Something new here? I get the same issue ... "/etc/rc.local ... can't cd to project_alias"

honk0504, 2019-03-30:
error found! It has to be the complete path ...
`cd /home/pi/project_alias && python3 app.py &` in rc.local

closed 2020-01-05 by torekndsn