bjoernkarmann/project_alias #24: Microphone issues

sempei13, 2019-03-09:
For reasons I don't get, the alias isn't registering any sound.  I've used this same hardware (exactly the same, not even a duplicate) before and got it working.  I thought that maybe the microphone was registering as from my webcam (which I had initially installed), but even with it connected and the pi rebooted I get nothing.

It's an AIY kit and the Google assistant part works great (so I know the integrated mic does work).  I just don't know how to tell alias to listen to THAT mic.

My plan is to kill the interference sound (file with silence?!) and instead of playing the "ok google" file, have it either use a relay to short the two wires that connected to the original button OR there's talk that I could send a signal to those pins internally and use the relay for other stuff.

Any ideas on those fronts, too?

sempei13, 2019-03-10:
Reinstalled and that might have fixed it.

sempei13, 2019-03-10:
Okay, this is odd.  It was looking for sound when I did the initial test before changing the lines in the ai.py and globals.py files.  Then, it stopped.  It registers I'm holding down the button, but won't record any samples.  Hmm.  More work to do tomorrow, I guess.

sempei13, 2019-03-19:
So to be clear, it's not working.  I don't know why because it worked before with this same hardware.

sempei13, 2019-04-24:
Still nothing.  It doesn't matter whether I autostart it or not.  It doesn't recognize the microphone.

sresam89, 2019-08-10:
I tried recording an audio through the microphone and played it back, seems like there is a ton of noise(buzzzzz ) in the recorded file. i think there is an issue with the Remic kit. Will try with google aiy kit and update the thread

bjoernkarmann, 2020-01-05:
hm, that does sound strange. We have completely re-written the code for version 2.0, including a different sound architecture, so maybe try if the 2.0 software fixes your issue.

TheStigh, 2020-02-21:
> 
> 
> hm, that does sound strange. We have completely re-written the code for version 2.0, including a different sound architecture, so maybe try if the 2.0 software fixes your issue.

Hi @bjoernkarmann bumping this active - I don't have the `ReSpeaker 2-Mics Pi HAT`but I do have the Google AIY Voice HAT and Voice HAT Microphone - should this work 'out-of-the-box'?

Also, what about not running any Voice HAT but the 'unofficial' RPi USB Mic ?

closed 2020-03-03 by torekndsn