bjoernkarmann/project_alias #26: Help wanted: I need to count triggering "Keyword"

lord-x-type, 2019-03-10:
Hi everyone,
i have a short question:
I would like to count how often Alias is triggering in 24h, without saying my keyword.
I thought about something like this: python3 app.py >> alias.log
and then load the log in text editor and use the search function to list and count the trigger events.
What would be this "text" to search for?

Or is there an other .. smarter possibility to get the trigger count for my settings?
I just want to check my settings..
Thank you very much.

philg1234, 2019-09-29:
hi lord-x-type,

have you already found a solution to spend the number of triggers?

This would interest me too.

lord-x-type, 2019-10-13:
Hi,
Sorry for the late reply.
I stopped using this project a while ago.
Was not realy happy with it. 
Sorry that i can not help you further.
Greetings

bjoernkarmann, 2020-01-05:
Hi,
This is not an issue but, but is sort of addressed in version 2.0. You can now see what trigger word was activated in the interface. This same communication should be easy to log with a simple print command.

closed 2020-01-05 by bjoernkarmann