bjoernkarmann/project_alias #28: Support for ReSpeaker 4-Mic Array

dewenni, 2019-04-17:
I found that there is another ReSpeaker board that looks interesting for this project.
It comes with 4 instead of 2 mics that might improve the voice recognition.
And it has also an LED ring that could be used to signalize that alias is listening.

http://wiki.seeedstudio.com/ReSpeaker_4_Mic_Array_for_Raspberry_Pi/

It would be nice if this will be also supported in future.
What do you think?

nitrag, 2019-06-07:
Bump?

https://www.amazon.com/seeed-Studio-ReSpeaker-4-Mic-Raspberry/dp/B076SSR1W1/ref=sr_1_10?keywords=respeaker&qid=1559932656&s=gateway&sr=8-10

bjoernkarmann, 2020-01-05:
This would be very possible to use! But the 3D files for Alias will currently not work with this ReSpeaker. If you like to modify the 3D model please do, but we do not have any plans of doing so. 
We have just released version 2.0, it has much better voice recognition! Thanks

closed 2020-01-05 by bjoernkarmann