bjoernkarmann/project_alias #29: Copy the ssh and wpa_supplicant.conf files

aro78, 2019-05-02:
Hello everyone,

i am quiete new in the raspberry world (only successfull experience with the Pi hole Project)

I stuck at the Point "2. Copy the ssh and wpa_supplicant.conf files from the setup folder to the SD card (boot)"
I copy it as  root to the /boot folder but after reboot both files are gone.
other folders like documents are o.k. after copieing, the poroblem is only in boot folder 
I tried two sd cards and also copy from usb stick to the card.
Nothinmg works and i cant find a solution in the web.

Thanks for helping.

Kindest Regards Andreas

siperdist, 2019-05-02:
Hi Andreas

this is not an issue. If you typed in your ssid and psk correctly  then your  pi should connect to your WiFi. Do not remove "" before and after the ssid and psk.

Florian

siperdist, 2019-05-02:
> Hello everyone,
> i am quiete new in the raspberry world (only successfull experience with the Pi hole Project)
> I stuck at the Point "2. Copy the ssh and wpa_supplicant.conf files from the setup [...]

Hi Andreas

this is not an issue. If you typed in your ssid and psk correctly then your pi should connect to your WiFi. Do not remove "" before and after the ssid and psk.

Florian

aro78, 2019-05-03:
Hi Florian,

thanks for helping.
Yea the "" were missing and i have not moved forward, because of the missing files.
Now the raspi finds the WLAN, but the 2 files are still gone.
Anyway. I wil go on and try to finish the project at my Alexa.
I am waiting the delivery of the mic unit and the speakers and then i will move forward.
Kindest Regards
Andreas

siperdist, 2019-05-03:
> Hi Florian,
> thanks for helping.
> Yea the "" were missing and i have not moved forward, because of the missing files.
> Now the raspi finds the WLAN, but the 2 files are still gone.
> Anyway. I wil go on [...]

Hi,

you don't need the files anymore, they're only to activate ssh and to set the WiFi config. 

Florian

aro78, 2019-05-03:
O.k. i understand. Thank you very much
BR
Andreas

warriar, 2019-06-22:
fixed, wasnt an issue - could be closed

closed 2019-09-29 by torekndsn