bjoernkarmann/project_alias #32: Installation issues

sempei13, 2019-06-06:
When I run "sudo pip3 install tensorflow keras" I have trouble with Scipy.  Here's the message I get 
`Command "/usr/bin/python3 -u -c "import setuptools, tokenize;__file__='/tmp/pip-build-gisif9_e/scipy/setup.py';f=getattr(tokenize, 'open', open)(__file__);code=f.read().replace('\r\n', '\n');f.close();exec(compile(code, __file__, 'exec'))" install --record /tmp/pip-w_ck2k79-record/install-record.txt --single-version-externally-managed --compile" failed with error code 1 in /tmp/pip-build-gisif9_e/scipy/`

So, when I try to run app.py, it doesn't work...at all.  This is the message I get when I try to run it manually: ` File "app.py", line 8, in <module>
    from modules import connection
  File "/home/pi/project_alias/modules/connection.py", line 7, in <module>
    from modules import sound
  File "/home/pi/project_alias/modules/sound.py", line 5, in <module>
    from python_speech_features import mfcc
  File "/usr/local/lib/python3.5/dist-packages/python_speech_features/__init__.py", line 1, in <module>
    from .base import *
  File "/usr/local/lib/python3.5/dist-packages/python_speech_features/base.py", line 6, in <module>
    from scipy.fftpack import dct
ImportError: No module named 'scipy'`

It loads on another SD card that I loaded it on earlier, but I've tried two versions of the OS (the one from AIY, from Google, and a regular, and Raspbian stretch full from April 2019.)

My only two thoughts are either that the newest version of Raspbian don't like it or that something is going on with the source files.  It is a Class4 16 gb card vs. a Class 10 32 gb, but I don't think that's it.

Any ideas?

PaulSonOfLars, 2019-06-08:
The second error says you haven't got scipy installed. Try installing that using:
```
sudo pip3 install scipy
```

You also __need__ to have keras for this project to work, so if your installation of that fails, you'll have to fix that first. I'm sure theres more information regarding the failure, would you mind providing more output?

sempei13, 2019-06-08:
I started the install of scipy by itself.  Here's the current status of things: `ollecting scipy
  Using cached https://files.pythonhosted.org/packages/cb/97/361c8c6ceb3eb765371a702ea873ff2fe112fa40073e7d2b8199db8eb56e/scipy-1.3.0.tar.gz
Building wheels for collected packages: scipy
  Running setup.py bdist_wheel for scipy ... \`

If history repeats itself, it will stay there (the "\" isn't moving, as it did when it first started), and then I'll see that there was a problem.  I'll update the post with what it says when it finishes.

warriar, 2019-06-22:
Same issue as #33 - for me it works after a couple of tries chaning nothing.

Just retry: 
pip3 install --no-cache-dir tensorflow keras 
until it works....

sempei13, 2019-07-01:
I just tried these all again and didn't get a hash problem so I thought that maybe it worked.  Then I tried  running the alias python 3 app and got: "23.0
Using TensorFlow backend.
/usr/lib/python3.5/importlib/_bootstrap.py:222: RuntimeWarning: compiletime version 3.4 of module 'tensorflow.python.framework.fast_tensor_util' does not match runtime version 3.5
  return f(*args, **kwds)
/usr/lib/python3.5/importlib/_bootstrap.py:222: RuntimeWarning: builtins.type size changed, may indicate binary incompatibility. Expected 432, got 412
  return f(*args, **kwds)
Invalid card number.
Usage: amixer <options> [command]

Available options:
  -h,--help       this help
  -c,--card N     select the card
  -D,--device N   select the device, default 'default'
  -d,--debug      debug mode
  -n,--nocheck    do not perform range checking
  -v,--version    print version of this program
  -q,--quiet      be quiet
  -i,--inactive   show also inactive controls
  -a,--abstract L select abstraction level (none or basic)
  -s,--stdin      Read and execute commands from stdin sequentially
  -R,--raw-volume Use the raw value (default)
  -M,--mapped-volume Use the mapped volume

Available commands:
  scontrols       show all mixer simple controls
  scontents	  show contents of all mixer simple controls (default command)
  sset sID P      set contents for one mixer simple control
  sget sID        get contents for one mixer simple control
  controls        show all controls for given card
  contents        show contents of all controls for given card
  cset cID P      set control contents for one control
  cget cID        get control contents for one control
Expression 'alsa_snd_pcm_hw_params_set_period_size_near( pcm, hwParams, &alsaPeriodFrames, &dir )' failed in 'src/hostapi/alsa/pa_linux_alsa.c', line: 924
Expression 'alsa_snd_pcm_hw_params_set_period_size_near( pcm, hwParams, &alsaPeriodFrames, &dir )' failed in 'src/hostapi/alsa/pa_linux_alsa.c', line: 924
ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.front
ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.rear
ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.center_lfe
ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.side
ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.surround21
ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.surround21
ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.surround40
ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.surround41
ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.surround50
ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.surround51
ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.surround71
ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.iec958
ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.iec958
ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.iec958
ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.hdmi
ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.hdmi
ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.modem
ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.modem
ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.phoneline
ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.phoneline
ALSA lib confmisc.c:1281:(snd_func_refer) Unable to find definition 'defaults.bluealsa.device'
ALSA lib conf.c:4528:(_snd_config_evaluate) function snd_func_refer returned error: No such file or directory
ALSA lib conf.c:4996:(snd_config_expand) Args evaluate error: No such file or directory
ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM bluealsa
ALSA lib confmisc.c:1281:(snd_func_refer) Unable to find definition 'defaults.bluealsa.device'
ALSA lib conf.c:4528:(_snd_config_evaluate) function snd_func_refer returned error: No such file or directory
ALSA lib conf.c:4996:(snd_config_expand) Args evaluate error: No such file or directory
ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM bluealsa
Expression 'alsa_snd_pcm_hw_params_set_period_size_near( pcm, hwParams, &alsaPeriodFrames, &dir )' failed in 'src/hostapi/alsa/pa_linux_alsa.c', line: 924
ALSA lib pcm_dmix.c:1052:(snd_pcm_dmix_open) unable to open slave
Cannot connect to server socket err = No such file or directory
Cannot connect to server request channel
jack server is not running or cannot be started
JackShmReadWritePtr::~JackShmReadWritePtr - Init not done for -1, skipping unlock
JackShmReadWritePtr::~JackShmReadWritePtr - Init not done for -1, skipping unlock
Traceback (most recent call last):
  File "app.py", line 155, in <module>
    stream = sound.initialize()
  File "/home/pi/project_alias/modules/sound.py", line 39, in initialize
    stream_callback=audio_callback)
  File "/usr/lib/python3/dist-packages/pyaudio.py", line 750, in open
    stream = Stream(self, *args, **kwargs)
  File "/usr/lib/python3/dist-packages/pyaudio.py", line 441, in __init__
    self._stream = pa.open(**arguments)
OSError: [Errno -9996] Invalid input device (no default output device)"

warriar, 2019-07-01:
@sempei13 install worked nowbits not finding the audiocard. Verify the audio driver setup and make sure you see the shield in alsamixer

sresam89, 2019-08-09:
> The second error says you haven't got scipy installed. Try installing that using:
>
> ```
> sudo pip3 install scipy
> ```
>
> You also **need** to have keras for this project to work, so if your installation [...]

I tried initially
`sudo pip3 install scipy` 
I was getting a random build error some posts suggested to try
`sudo pip3 install scipy --no-cache-dir`
and then I started getting
`error extension pymc flib has fortran sources but no fortran compiler found`

Then I had do 
`sudo apt-get install gfortran; sudo pip3 install scipy --no-cache-dir`
and wait forever to build and install it (so be patient) and to make it work

bjoernkarmann, 2020-01-05:
With version 2.0 the install has become much easier. This will solve the issue!
I hope it works out, otherwise let me know.

closed 2020-01-05 by bjoernkarmann