bjoernkarmann/project_alias #33: can't install Keras

mrbjoernson, 2019-06-11:
Hello, i can't install keras. 

if i want to install keras, after a long time nothing happens and its freezes. 

Does somebody has any idea?

`pi@raspberrypi:~ $ pip3 install --no-cache-dir keras
Collecting keras
  Downloading https://files.pythonhosted.org/packages/5e/10/aa32dad071ce52b5502266b5c659451cfd6ffcbf14e6c8c4f16c0ff5aaab/Keras-2.2.4-py2.py3-none-any.whl (312kB)
    100% |████████████████████████████████| 317kB 4.0MB/s 
Collecting keras-preprocessing>=1.0.5 (from keras)
  Downloading https://files.pythonhosted.org/packages/28/6a/8c1f62c37212d9fc441a7e26736df51ce6f0e38455816445471f10da4f0a/Keras_Preprocessing-1.1.0-py2.py3-none-any.whl (41kB)
    100% |████████████████████████████████| 51kB 4.7MB/s 
Collecting scipy>=0.14 (from keras)
  Downloading https://files.pythonhosted.org/packages/cb/97/361c8c6ceb3eb765371a702ea873ff2fe112fa40073e7d2b8199db8eb56e/scipy-1.3.0.tar.gz (23.6MB)
    100% |████████████████████████████████| 23.6MB 3.9MB/s 
Collecting h5py (from keras)
  Downloading https://www.piwheels.org/simple/h5py/h5py-2.9.0-cp35-cp35m-linux_armv7l.whl (3.5MB)
    100% |████████████████████████████████| 3.5MB 7.0MB/s 
Collecting six>=1.9.0 (from keras)
  Downloading https://files.pythonhosted.org/packages/73/fb/00a976f728d0d1fecfe898238ce23f502a721c0ac0ecfedb80e0d88c64e9/six-1.12.0-py2.py3-none-any.whl
Collecting keras-applications>=1.0.6 (from keras)
  Downloading https://files.pythonhosted.org/packages/71/e3/19762fdfc62877ae9102edf6342d71b28fbfd9dea3d2f96a882ce099b03f/Keras_Applications-1.0.8-py3-none-any.whl (50kB)
    100% |████████████████████████████████| 51kB 7.0MB/s 
Collecting pyyaml (from keras)
  Downloading https://files.pythonhosted.org/packages/a3/65/837fefac7475963d1eccf4aa684c23b95aa6c1d033a2c5965ccb11e22623/PyYAML-5.1.1.tar.gz (274kB)
    100% |████████████████████████████████| 276kB 4.6MB/s 
Collecting numpy>=1.9.1 (from keras)
  Downloading https://files.pythonhosted.org/packages/d3/4b/f9f4b96c0b1ba43d28a5bdc4b64f0b9d3fbcf31313a51bc766942866a7c7/numpy-1.16.4.zip (5.1MB)
    100% |████████████████████████████████| 5.1MB 6.3MB/s 
Installing collected packages: six, numpy, keras-preprocessing, scipy, h5py, keras-applications, pyyaml, keras
  Running setup.py install for numpy ... done
  Running setup.py install for scipy ... \packet_write_wait: Connection to 2002:5b43:6398::c87d:5754:ffa2:5f3 port 22: Broken pipe`

mrbjoernson, 2019-06-12:
probably the main problem is scipy?!

warriar, 2019-06-22:
I have the same problem when doing 

pip3 install --no-cache-dir tensorflow keras

However i can eventuall install it when i try it a couple of times.
Sounds stupid but it works eventually.

I would say the pip3 source servers have some kind of issue sending broken packages some of the time?!

sempei13, 2019-07-01:
I got Scipy to install by running the script separately, but that was after follwing Warriar's advice and doing "pip3 install --no-cache-dir tensorflow keras" (without the quotes, of course) multiple times.  It keeps telling me that there's a problem with the Hash, that it doesn't match.  When I do the normal install stript, it sticks and then hours later quits.

sempei13, 2019-07-01:
Here's the error I get: "THESE PACKAGES DO NOT MATCH THE HASHES FROM THE REQUIREMENTS FILE. If you have updated the package versions, please update the hashes. Otherwise, examine the package contents carefully; someone may have tampered with them."  Then it lists the hashes.

torekndsn, 2020-01-05:
Hi, 
We just released Alias v.2 which now is the main branch. We have been focusing on getting that ready for some time now, as it is more reliable, easier to install, and has new features! 

We hope you wanna give it a shot! 
Best, Tore

closed 2020-01-05 by torekndsn