bjoernkarmann/project_alias #38: TensorFlow - Assertion `ret == self->nfds' failed

lle, 2019-09-09:
Had to install TensorFlow with the flag "--no-cache-dir", not sure if this is why things are not working out. Here is the problem:
Upon starting "app.py", the following warning appears and the processs is aborted.
Unsure how to proceed next.

`WARNING:tensorflow:From /usr/local/lib/python3.7/dist-packages/keras/backend/tensorflow_backend.py:4267: The name tf.nn.max_pool is deprecated. Please use tf.nn.max_pool2d instead.

WARNING:tensorflow:From /usr/local/lib/python3.7/dist-packages/keras/backend/tensorflow_backend.py:3733: calling dropout (from tensorflow.python.ops.nn_ops) with keep_prob is deprecated and will be removed in a future version.
Instructions for updating:
Please use `rate` instead of `keep_prob`. Rate should be set to `rate = 1 - keep_prob`.
WARNING:tensorflow:From /usr/local/lib/python3.7/dist-packages/keras/optimizers.py:793: The name tf.train.Optimizer is deprecated. Please use tf.compat.v1.train.Optimizer instead.

WARNING:tensorflow:From /usr/local/lib/python3.7/dist-packages/tensorflow_core/python/ops/nn_impl.py:182: add_dispatch_support.<locals>.wrapper (from tensorflow.python.ops.array_ops) is deprecated and will be removed in a future version.
Instructions for updating:
Use tf.where in 2.0, which has the same broadcast rule as np.where
python3: src/hostapi/alsa/pa_linux_alsa.c:3641: PaAlsaStreamComponent_BeginPolling: Assertion `ret == self->nfds' failed.
Aborted
`

notimp, 2019-09-12:
Same issue here.

The project development seems to have moved over to version 2.0 of the software (see branch), although I cant get that to run either. It has different dependencies, no documentation, ...

In the readme for that project they mention 'simply download our easy to install disk image" - and provide a link that 404's for anyone thats not the project lead presumably. In the readme they indicate, that 2.0 has replaced thee normal master - which isn't the case, and provide a link back to the 'old version' which simply links back to a generic product page.

This project is broken as of now. Hope the leads still attend to it and get it to a working state again. Its not like I bought all the hardware, because of a PR tour they are currently engaged in (all kinds of outlets pick up the story to this ultimately broken project).

ataboo, 2019-09-16:
In trying to get 2.0 working, I installed the following through apt:
- apulse-dev
- libasound2-dev
- mplayer

I went back to the master branch and it managed to run.

I also installed pocketsphinx and swig but I think it was apulse-dev or libasound2-dev that fixed it.

lle, 2019-09-16:
Great. I'll try that tonight

notimp, 2019-09-23:
Successfully installed and ran Project Alias for the first time (master branch - today).
Step by step:

Use the previous version of Raspian (stretch) you can get from this link:
http://downloads.raspberrypi.org/raspbian/images/raspbian-2019-04-09/
It comes with Python 3.5.3 and will remain on that version even after apt-get update and apt-get upgrade.

Follow the tutorial for the rest except for installing tensorflow and keras.
On my first try I installed tensorflow 1.8.0 from here ( https://github.com/lhelontra/tensorflow-on-arm/releases  cp35-none-linux_armv7l version) wget ant then sudo pip3 install ./tensorflow-1.8.0-cp35-none-linux_armv7l.whl , the auto installed keras then errored out - so I had to uninstall it (sudo pip3 uninstall keras), and then install an older version (tried 2.1.6 and it 'worked') with sudo pip3 install keras==2.1.6

For the rest of the dependancies just follow the tutorial. Read the other issue threads, if you've got an issue with autolaunching app.py after boot.

I bet this would also work with 

sudo pip3 install --no-cache-dir tensorflow
and
sudo pip 3 install keras

so the current versions. But I havent tested that yet.

So all you need is an older version of raspbian, and maybe - an older version of tensorflow and keras -- then the entire tutoral works again. ;) (as of today)

(Read this: http://wiki.seeedstudio.com/ReSpeaker_2_Mics_Pi_HAT/ and use alsamixer for setting the speaker and mic level. :) )

philg1234, 2019-09-24:
Thanks. 

I'll test it tonight and give feedback.

notimp, 2019-09-24:
Two more things. 

sudo reboot 

directly after installing the ReSpeaker drivers, because the installation routine will tell you so. Also you will be tempted to start Project Alias after installing the other dependencies without a reboot. :)

The 'url' to surf by on your phone in my case turned out to be raspberrypi:5050 and not raspberrypi.local:5050

Thats it. (Updated the previous posting for clarity (and a slightly newer version of keras (2.2.0. first one posted as working was 2.1.6).)

philg1234, 2019-09-24:
@notimp 

I tried to download the raspbian version you linked to.

However, he tells me that the file is damaged and I can not unzip it

philg1234, 2019-09-24:
Short supplement.

I have just learned the first alias.

However, the alias triggers too often without ihc saying anything.

I have the test unplugged the speakers and still the alias is triggered: D.

Where could the mistake lie here?

notimp, 2019-09-25:
Same issue here. You can set the microphone (and speaker level) using alsamixer.

(Read this: http://wiki.seeedstudio.com/ReSpeaker_2_Mics_Pi_HAT/ and use alsamixer for setting the speaker and mic level. :) )

Havent played around with - if thats permanent, or how to make it permanent - they also mention in the readme to edit a line to raise speaker levels for alexa.

What I've found due to some limited testing is, that if you set mic level to something that according to the ReSpeaker test routine (put on headphones, and listen to what you are speaking and ReSpeaker is picking op (for line to script this, see link above)) -- still is perfectly normal, and in fact minimizes the pick up of microphone noise -- according to the html/javascript indicator on your phone - has the project alias not picking up "sufficient sound" anymore.

So - either the project in this state by far isn't as workable as the youtube video indicated - or maybe we are dealing with unexpected issues, due to dependency versions, or inter unit variability on the ReSpeaker hats... :)

Hard to tell. ;) Try raising Tensorflow and Keras versions slowly and in correspondance to the version that was releast of the other, when a new version of the former came out.

Try to get version 2.0 of the project running.

Blame it on the dev. Those are your potential options. ;)

But I can confirm, that I have the oversensitivity on part of it triggering OK google speech samples as well. Thats not just you. :)

philg1234, 2019-09-25:
Thanks for your support.

I think it's a shame that the project does not work as it did in the video.

Presumably, this was not thought to end properly, as often as me the Project ALias antriggert.

Tonight I will again try to record several aliases, as well as some background noise.

I would love to get this project working, as I find it pretty cool personally.

Maybe it will come out in the near future, the linked 2.0, with which it works better.

I give nochmasl feedback.

philg1234, 2019-09-28:
Today I have tried again a lot to get the project alias a new name.

However, it does not work.

I reset everything via "Reset Alias" and initially learned the alias without any background sound.

As soon as I have learned this, alias will start every 2 seconds without anything being said.

When I learn the alias with the background sound ON, the alias does not start, no matter what is said.

Does this project work for anyone?

Does anyone know about the Verison 2.0?

Greetings Phil

janniks, 2019-10-14:
@notimp can you share your pip dependencies? `pip3 freeze > requirements.txt`

ataboo, 2019-10-24:
Following @notimp's instructions got it running for me, but it seems to activate erratically as others are reporting.

Here's my dependencies:
>absl-py==0.8.1
astor==0.8.0
automationhat==0.2.0
bleach==1.5.0
blinker==1.3
blinkt==0.1.2
buttonshim==0.0.2
Cap1xxx==0.1.3
chardet==2.3.0
click==6.6
colorama==0.3.7
colorzero==1.1
cryptography==1.7.1
drumhat==0.1.0
envirophat==1.0.0
ExplorerHAT==0.4.2
Flask==0.12.1
Flask-SocketIO==4.2.1
fourletterphat==0.1.0
gast==0.3.2
gpiozero==1.5.0
grpcio==1.24.3
h5py==2.10.0
html5lib==0.9999999
idna==2.2
itsdangerous==0.24
Jinja2==2.8
Keras==2.2.0
Keras-Applications==1.0.2
Keras-Preprocessing==1.0.1
keyring==10.1
keyrings.alt==1.3
Markdown==3.1.1
MarkupSafe==0.23
microdotphat==0.2.1
mote==0.0.4
motephat==0.0.2
numpy==1.17.3
oauthlib==2.0.1
pantilthat==0.0.7
pgzero==1.2
phatbeat==0.1.1
pianohat==0.1.0
picamera==1.13
picraft==1.0
piglow==1.2.4
pigpio==1.38
Pillow==4.0.0
protobuf==3.10.0
pyasn1==0.1.9
PyAudio==0.2.11
pycrypto==2.6.1
pygame==1.9.3
pygobject==3.22.0
pyinotify==0.9.6
PyJWT==1.4.2
pyOpenSSL==16.2.0
pyserial==3.2.1
python-apt==1.1.0b5
python-engineio==3.10.0
python-socketio==4.3.1
python-speech-features==0.6
pyxdg==0.25
PyYAML==5.1.2
rainbowhat==0.1.0
requests==2.12.4
requests-oauthlib==0.7.0
RPi.GPIO==0.6.5
RTIMULib==7.2.1
scipy==1.3.1
scrollphat==0.0.7
scrollphathd==1.2.1
SecretStorage==2.3.1
sense-hat==2.2.0
simplejson==3.10.0
six==1.12.0
skywriter==0.0.7
sn3218==1.2.7
spidev==3.3
ssh-import-id==5.6
tensorboard==1.8.0
tensorflow==1.8.0
termcolor==1.1.0
touchphat==0.0.1
twython==3.4.0
unicornhathd==0.0.4
urllib3==1.19.1
Werkzeug==0.11.15

torekndsn, 2020-01-05:
Hi,
We are glad to announce Alias 2.0 is now available!! We are well aware that this comes very late and has been requested by many who struggled to install or even calibrate Alias. We are sorry about that! Nevertheless, we are proud of Release 2.0 and have already had a lot of fun with it! We have introduced a bunch of new features that give even more freedom, and the installation process is now made into a single image file.

Best, Tore

closed 2020-01-05 by torekndsn