bjoernkarmann/project_alias #40: Training alias not possible

philg1234, 2019-09-22:
Hi,

I have a problem calling the page "raspberrypi.local: 5050".

The site is not accessible.
I have everything installed and configured according to the instructions.
But its not work.

if I enter the command "cd project_alias && python3 app.py &" over ssh and putty the message comes the follow:

> pi@raspberrypi:~ $ cd project_alias && python3 app.py &
> [1] 1201
> pi@raspberrypi:~ $ pygame 1.9.4.post1
> Hello from the pygame community. https://www.pygame.org/contribute.html
> 23.0
> Using TensorFlow backend.
> Traceback (most recent call last):
>   File "app.py", line 10, in <module>
>     from modules import ai
>   File "/home/pi/project_alias/modules/ai.py", line 6, in <module>
>     import keras
>   File "/usr/local/lib/python3.7/dist-packages/keras/__init__.py", line 3, in <m        odule>
>     from . import utils
>   File "/usr/local/lib/python3.7/dist-packages/keras/utils/__init__.py", line 6,         in <module>
>     from . import conv_utils
>   File "/usr/local/lib/python3.7/dist-packages/keras/utils/conv_utils.py", line         9, in <module>
>     from .. import backend as K
>   File "/usr/local/lib/python3.7/dist-packages/keras/backend/__init__.py", line         1, in <module>
>     from .load_backend import epsilon
>   File "/usr/local/lib/python3.7/dist-packages/keras/backend/load_backend.py", l        ine 90, in <module>
>     from .tensorflow_backend import *
>   File "/usr/local/lib/python3.7/dist-packages/keras/backend/tensorflow_backend.        py", line 5, in <module>
>     import tensorflow as tf
> ModuleNotFoundError: No module named 'tensorflow'

can someone help me with this?

Thank you

greetings Phil

sknadig, 2019-09-24:
Try this
python3 -m pip install tensorflow

philg1234, 2019-09-24:
I have already missed that, but without success.

I saw in another thread # 38 that somebody got the project running with older versions.

This is what I will try tonight.

philg1234, 2019-09-25:
Update:

I have now performed a reinstallation as described in Thread # 38. After that, everything worked fine and the page can be called correctly. 

Gretings Phil

closed 2019-09-25 by philg1234