bjoernkarmann/project_alias #42: Memory consumption of app.py

SargonofAssyria, 2019-10-27:
I am using: Linux  4.19.75-v7+ #1270 SMP Tue Sep 24 18:45:11 BST 2019 armv7l GNU/Linux (Buster)
Python version: 3.7.3 on a Rapberry Pi 3 model A+ The system is close to a default Buster install with only a few extra's.
Everything is working without errors, except that app.py is consuming 300-500 kBytes more every minute. Is starts with approx. 38 MB en run until approx. 75-80 MB and then sometimes the system crashes and sometimes only the app.py crashes.
Anyone any idea how to prevent this?

seasick, 2019-10-30:
See PR #37 for a possible fix

torekndsn, 2020-01-05:
Hi, 

Version 2 is now out and is based on much more reliable software than version 1. 
The installation is also simplified down to a single image file to flash an sd card with! 

Best, Tore

closed 2020-01-05 by torekndsn