bjoernkarmann/project_alias #45: Installation failed

idem2lyon, 2019-12-21:
I try to execute the readme.
It failed.
I try to follow [Issues 32](https://github.com/bjoernkarmann/project_alias/issues/32) and still doesn't work.

`Command "/usr/bin/python3 -u -c "import setuptools, tokenize;__file__='/tmp/pip-install-7yz1q1em/scipy/setup.py';f=getattr(tokenize, 'open', open)(__file__);code=f.read().replace('\r\n', '\n');f.close();exec(compile(code, __file__, 'exec'))" install --record /tmp/pip-record-v3js5trq/install-record.txt --single-version-externally-managed --compile" failed with error code 1 in /tmp/pip-install-7yz1q1em/scipy/`

This project is dead ?

martinge17, 2019-12-21:
i dont think it

martinge17, 2019-12-21:
I hope that the devs will soon help us

idem2lyon, 2019-12-22:
I hope too....

ebk46, 2020-01-05:
I believe if you update setuptools it will fix that error but there are loads more, and honestly even after getting it running, it doesn't seem to work...

torekndsn, 2020-01-05:
Hi, we are glad to announce Alias 2.0 is now available!! We are well aware that this comes very late and has been requested by many who struggled to install or even calibrate Alias. We are sorry about that!

Nevertheless, we are proud of Release 2.0 and have already had a lot of fun with it! We have introduced a bunch of new features that give even more freedom, and the installation process is now made into a single image file.

Best, Tore

closed 2020-01-05 by torekndsn