bjoernkarmann/project_alias #47: Readback is not correct

siperdist, 2020-01-06:
Good morning,

I installed 2.0 yesterday but my readback is not correct. I use "Alias" to wake Alias up, when I speak the command "Alias wie viel Uhr ist es?" The noice is canceled and I can hear the readback "ok google" and my gogle home wakes up. Unfortunately the rest  is not send to my google home. Am I doing something wrong?

Florian

torekndsn, 2020-01-06:
Hi Florian, 
Glad to hear that you already got 2.0 installed! 
So before you can say "wie viel Uhr ist es" your google home needs to be awake. The sequence of events should be: [You: "Alias"] - [Alias stops noise] - [Alias whispers: "Ok Google"] - [Alias lights up yellow] - [You say: "Alias wie viel Uhr ist es?"] So you need to hold a short pause between "Alias" and "wie viel Uhr ist es?"

I hope I understand the problem the right way, and that this helps! 
/Tore

siperdist, 2020-01-06:
Expected something like this, will try when I'm back home. Thanks for the super fast support.

siperdist, 2020-01-06:
Hi Tore,

thanks again for your support. I'm home now and I tried what you told me. I figured out the main problem isn't the missing readback (which I thought first) but that my google home isn't activated by "ok google" coming from the two speakers. I tried different languages and loudness settings. I can only activate it by my voice. The sound of the activation voice is a bit like from the 80's. Any suggestions?

Florian

torekndsn, 2020-01-06:
I can come up with 3 things to try: 

1. Make sure that the speakers are placed directly on top of the two microphones on the Google home, and that you can hear the "Ok, google" on activation if you listen closely. 

2. Make sure that Alias is actually activated and that you give it time to activate the Google Home. A yellow light emits from Alias when its activated and only after that is the assistant ready to listen. Try waiting 1 second after you hear the "Ok Google" before you ask for something. 

3. This is a long-shot, but try to change the gender of Alias. That changes the voice slightly, but it shouldn't be a problem as we have used both voices on many different Google Homes.

alefnode, 2020-04-04:
Same problem for me, so i fix it with that

https://github.com/bjoernkarmann/project_alias/pull/56