bjoernkarmann/project_alias #49: feedback Project Alias 2.0

idem2lyon, 2020-01-10:
Hi,
I've tested the Project Alias 2.0 yesterday.
If you want to connect using the url alias.local you need to have Bonjour...it"s not my case, so use IP address instead : 192.168.4.1
So, I configure Alias, set a new name, male voice, french, and test.... nothing append.
Google still working (I try changing volume, and check Noise enable) .. :(
And sometimes, I can ear my voice in speakers.

What's wrong with my Alias?

torekndsn, 2020-03-03:
Hi @idem2lyon 

So the wake word needs to be existing in the dictionary of the Pocketsphix library, which is English. So while you can change the language of what Alias says to the Assistant, the wake-word needs to be English. You can test with a more common word and increase the sensitivity of your wake word until it responds. 

Best, Tore

idem2lyon, 2020-03-17:
I use default wake word and it still the same.