bjoernkarmann/project_alias #50: Bugs with wake-word text entry from mobile

ebk46, 2020-01-17:
When I use the UI in the web-app from my phone (Android, Pixel 2, Chrome browser) and try to type a keyword I have the following 2 issues:

1. When the keyboard opens, the text input moves up beyond the viewport so it's not visible.
2. If I try to type a multi-word wake-word (e.g. Hey Jarvis), it will reverse the order (e.g. Jarvis Hey). I assume the cursor is moving to the beginning of the input when I press space for some reason, but because of issue #1 it's hard to tell.

TheStigh, 2020-02-29:
I have the same issue - using the Final2.0 image and exact identical HW as suggested.

johannespuchinger, 2020-03-01:
Had the same issue. I've switched the browser to desktop mode, then it worked.

TheStigh, 2020-03-01:
> 
> 
> Had the same issue. I've switched the browser to desktop mode, then it worked (issue 1)

Good idea :)

Have u tested the final2.0.img and/or the manual installation of v 2.0 ? Any success?

johannespuchinger, 2020-03-02:
I've downloaded the 2.0 img (SHA1: 4E78A1D8E67BAE40C6ABB922E51D7AF630DB6255) and I hear the noise (more a type of background music, people talking etc. - I used the 3.5mm jack on the HAT fpr testing), but have problems with the wakeword. 
Another problem is that the Pi and the ReSpeaker HAT will not fit in the 3d printed case - no chance.

torekndsn, 2020-03-03:
Thx for pointing out the Android issue. We have been testing the layout for desktop/IOS so we are not aware of how the layout works on android. 

@johannespuchinger 
if it's not responding to the wake word try increasing the sensitivity through the alias interface and remember to press the update button after changing. Most single words should be around 15-35 but for several words I've experimented with a sensitivity of up to 70 as well. Also, Alias can only be activated by words/Sentences that exist in the dictionary of the Pocketsphinx library that we are using, so try to test with a common word first. 

And for the reSpeaker hat fit, it might help to gently cut the corners of the reSpeaker board off. Not ideal I know. We will try to update the STL when possible. 

Best, Tore