bjoernkarmann/project_alias #52: the speakers do not give sound.

SEYDINALAYE, 2020-01-21:
Hello,

I just installed version 2.0 manually but I don't have the noise. In summary the speakers do not give sound.
Can you help me

compile:
pi@raspberrypi:~/project_alias $ python3 app.py
pygame 1.9.4.post1
Hello from the pygame community. https://www.pygame.org/contribute.html
starting socket thread

============================================
SERVER RUNNING ON: http://0.0.0.0:5050
============================================

play noise
 * Serving Flask app "modules.connect" (lazy loading)
 * Environment: production
   WARNING: Do not use the development server in a production environment.
   Use a production WSGI server instead.
 * Debug mode: off
amixer: Unable to find simple control 'Headphone',0

amixer: Unable to find simple control 'Speaker',0

init live speech
running speech

torekndsn, 2020-03-03:
Hey @SEYDINALAYE 

I don't know why it's not finding the soundcard but you can try to ssh into the raspberry after connecting to the Project Alias wifi: `ssh pi@alias.local`
And then bring up the Alsa mixer interface by typing: `alsamixer`
And then press F6 to get up the overview of available soundcards. 
Index 0 should be seed-2mic-voicecard. (see image)

What do you get there?

[image]

/Tore

alefnode, 2020-04-04:
If you are using jack on respeaker be careful to not use a jack with mic, for me that's produces an error when playing noise