bjoernkarmann/project_alias #54: Are there anybody that got the new final 2.0 image to work?

TheStigh, 2020-02-29:
Hi all, really sad there is no support from the dev's after releasing the 2.0.

I've got the exact same hw parts as they have used. Flashed the ready-made image. Got no noise/sound. There is absolutely no debugging or behavior info available.

Perhaps others here could assist each other?

ThePirat91, 2020-03-02:
Hello, I have the same problem as you.
I'm wondering if we are looking to a dead project....

torekndsn, 2020-03-03:
Hej there, 

It sounds like the sound card is not found. 
You can try debugging by connecting your computer to the Project Alias wifi and in your terminal ssh into the raspberry: `ssh pi@alias.local`

And then print the status of the service running alias: `sudo systemctl status alias.service`
On of the first lines should be "Simple mixer control 'Speaker',0"

You can also try typing `alsamixer` in the terminal and then press F6 to see available sound cards. it should be listed as the attached image: 

[image]

uNbAs, 2020-03-24:
I try to use v2.0 of projet alias but after reboot not have "seeed-2mic-voicecard" but I see in modules by lsmod:
snd_soc_wm8960         40960  0
snd_soc_ac108          53248  0
snd_soc_seeed_voicecard    16384  1 snd_soc_ac108
snd_soc_simple_card_utils    16384  1 snd_soc_seeed_voicecard
snd_soc_core          188416  5 snd_soc_seeed_voicecard,snd_soc_simple_card_utils,snd_soc_bcm2835_i2s,snd_soc_ac108,snd_soc_wm8960
snd_compress           20480  1 snd_soc_core
snd_pcm_dmaengine      16384  1 snd_soc_core
snd_pcm                98304  5 snd_pcm_dmaengine,snd_soc_bcm2835_i2s,snd_bcm2835,snd_soc_core,snd_soc_wm8960
snd_timer              32768  1 snd_pcm
snd                    73728  8 snd_compress,snd_timer,snd_bcm2835,snd_soc_core,snd_soc_wm8960,snd_pcm

[image]

alefnode, 2020-04-04:
Yeah, the image works with same hardware for me.

TheStigh, 2020-04-04:
Works for me also now :) I did a quite stupid mistake - connected the speaker to the WRONG output :( Though, it has a very slow response on the wake word and a lot of false hits. It can even be totally quiet in the living room and suddenly the lights light up !?!

Also, as reported in other issues the mobile view of the config is more or less useless until there is an update - either shift to Desktop view or use a laptop/pc to operate the admin page.

Closing the topic as the image are confirmed to work on identical hardware config.

closed 2020-04-04 by TheStigh