bjoernkarmann/project_alias #60: Pi4 compatibility issue

Rtabb2, 2020-06-13:
Good day and thank you for creating this project!

I've noticed that the "Easy Setup" doesn't work on a Pi4 (only get red light, doesn't seem to boot) but the manual setup goes okay until it comes time to train Alias. I can't seem to detect the WiFi network on any computer or phone. Do you know if this issue is specific to the Pi4, or am I missing something else?

fredddvargas, 2020-06-23:
I am going to try manual setup on the rpi4 once I have some free time and report back

holysoles, 2020-12-06:
@Rtabb2 the prebuilt image was likely built on a version of Raspbian that isn’t compatible with the Pi 4. Additionally, with version 2 of Project Alias there is not a training mode anymore.