bjoernkarmann/project_alias #64: Can't get to training mode

kw2583, 2020-09-19:
I've followed all of the setup instructions, and arrived at the step for training the wake word. When I attempt to connect to raspberrypi.local:5050, I get a DNS error. If I use <ip_address>:5050, I get to the page for setting the wake word. How can I get to the training page?

torekndsn, 2020-09-21:
Hej kw2583

If you connect to `http://alias.local` you should get to the alias app. And the training page was only part of version 1.0 In version 2.0 we are using a speech-to-text based way of recognizing the custom wake-up word. So you don't need to train. Just type in the word/name you would like to use, and calibrate the sensitivity if needed. 

/Tore

jaredfiacco2, 2021-01-22:
In case someone else has this issue - "http://alias.local" **worked on my laptop** browser, but it _didn't work on my phone._
I was using chrome on both devices. 
Really cool product! Thanks for making/maintaining it!